jmdmahdi/ESP32-USB-TO-BLE #2: does i need USB host

Ashish-Coder12, 2024-04-14:
i saw somewhere that i need a usb host to connect my usb keyboard to esp32 board because esp32 board does not understand usb connections

jmdmahdi, 2024-05-09:
Yep! ESP32-S2/S3 can handle USB connections all by themselves, no extra stuff needed ([ref](https://docs.espressif.com/projects/esp-faq/en/latest/software-framework/peripherals/usb.html)). For other ESP32s, you've got two choices: either add a USB Host controller module or use a software implementation on GPIOs. In this project, we've gone with the software option, using the [ESP32 USB Soft Host](https://github.com/tobozo/ESP32-USB-Soft-Host) Arduino library.

rakshitbharat, 2024-12-07:
> Yep! ESP32-S2/S3 can handle USB connections all by themselves, no extra stuff needed ([ref](https://docs.espressif.com/projects/esp-faq/en/latest/software-framework/peripherals/usb.html)). For other [...]

please update the readme with this instruction that esp32 will not out of box support for hardware based USB host 
many people are always be in confusion for it, usb host is needed to make any mechanical keyboard to work and so on.

Thanks

closed 2024-12-16 by rakshitbharat