jmdmahdi/ESP32-USB-TO-BLE #3: How to detect keyboard

dienvx1997bn, 2024-05-12:
Hello jmdmahdi,
I 've just tried to test your project with my ESP32 board. But i can not detect my keyboard.
here is the log
[image]

Could you make some short video to show how it works, It might help with me.

jmdmahdi, 2024-05-29:
Hello,

Make sure you have connected D+ and D- correctly to GPIO16 and GPIO17 of your ESP32 module. Sometimes, the pin numbers on the module itself are different from the GPIO numbers. Double-check the ESP32 pinout diagram for your specific board to ensure you have the correct connections.

Additionally, ensure your 5-volt power supply is rated high enough to power both the ESP32 and your keyboard. If you are using a separate power supply for the keyboard, the ground of the keyboard power supply must be connected to the ESP32's supply ground.

Could you send some pictures of your setup? It would help in diagnosing the problem.

rakshitbharat, 2024-12-03:
> Hello,
>
> Make sure you have connected D+ and D- correctly to GPIO16 and GPIO17 of your ESP32 module. Sometimes, the pin numbers on the module itself are different from the GPIO numbers. Double-check [...]

I have done the same what you said but it don't work on mechanical keyboard

closed 2024-12-16 by rakshitbharat