jmdmahdi/ESP32-USB-TO-BLE #4: Rewrite to make Bluetooth to BLE

comk22, 2024-05-28:
Hi,
I have a Bluetooth keyboard which I want to make accessible on several devices. I don't know much about the code you use but I would give it a shot by using chatgpt 4 to rewrite the code to receive the signals from my Bluetooth keyboard Do you think this is doable or is your code too far away from this idea?

jmdmahdi, 2024-05-29:
Hi,

For this project, we are sending keyboard data over BLE, so our current setup isn't designed to receive signals from a Bluetooth keyboard. However, I might be able to help you with your project. Please send me a message on Telegram so we can discuss it further.

Telegram ID: [@jmdmahdi](https://t.me/jmdmahdi)

Best regards.

rakshitbharat, 2024-12-02:
#14 try this PR

rakshitbharat, 2024-12-16:
latest source code have it with vscode platform io

closed 2024-12-16 by rakshitbharat