jmdmahdi/ESP32-USB-TO-BLE #8: There are many errors using these libraries

SaiRanga11, 2024-08-04:
In file included from /Users/sairanga/Documents/Arduino/ESP32-USB-TO-BLE/ESP32-USB-TO-BLE.ino:5:
/Users/sairanga/Documents/Arduino/libraries/ESP32-USB-TO-BLE-master/BleDevice.h:10:44: error: expected class-name before ',' token
 class BleDevice : public BLEServerCallbacks, public BLECharacteristicCallbacks {
                                            ^
/Users/sairanga/Documents/Arduino/libraries/ESP32-USB-TO-BLE-master/BleDevice.h:10:80: error: expected class-name before '{' token
 class BleDevice : public BLEServerCallbacks, public BLECharacteristicCallbacks {
                                                                                ^
/Users/sairanga/Documents/Arduino/libraries/ESP32-USB-TO-BLE-master/BleDevice.h:12:3: error: 'BLEHIDDevice' does not name a type; did you mean 'BleDevice'?
   BLEHIDDevice* hid;
   ^~~~~~~~~~~~
   BleDevice
/Users/sairanga/Documents/Arduino/libraries/ESP32-USB-TO-BLE-master/BleDevice.h:13:3: error: 'BLECharacteristic' does not name a type; did you mean 'Character_h'?
   BLECharacteristic* inputMouse;
   ^~~~~~~~~~~~~~~~~
   Character_h
/Users/sairanga/Documents/Arduino/libraries/ESP32-USB-TO-BLE-master/BleDevice.h:14:3: error: 'BLECharacteristic' does not name a type; did you mean 'Character_h'?
   BLECharacteristic* inputKeyboard;
   ^~~~~~~~~~~~~~~~~
   Character_h
/Users/sairanga/Documents/Arduino/libraries/ESP32-USB-TO-BLE-master/BleDevice.h:15:3: error: 'BLECharacteristic' does not name a type; did you mean 'Character_h'?
   BLECharacteristic* outputKeyboard;
   ^~~~~~~~~~~~~~~~~
   Character_h
/Users/sairanga/Documents/Arduino/libraries/ESP32-USB-TO-BLE-master/BleDevice.h:16:3: error: 'BLECharacteristic' does not name a type; did you mean 'Character_h'?
   BLECharacteristic* inputMediaKeys;
   ^~~~~~~~~~~~~~~~~
   Character_h
/Users/sairanga/Documents/Arduino/libraries/ESP32-USB-TO-BLE-master/BleDevice.h:17:3: error: 'BLEAdvertising' does not name a type
   BLEAdvertising* advertising;
   ^~~~~~~~~~~~~~
/Users/sairanga/Documents/Arduino/libraries/ESP32-USB-TO-BLE-master/BleDevice.h:31:26: error: 'BLEServer' has not been declared
   virtual void onConnect(BLEServer* pServer) override;
                          ^~~~~~~~~
/Users/sairanga/Documents/Arduino/libraries/ESP32-USB-TO-BLE-master/BleDevice.h:32:29: error: 'BLEServer' has not been declared
   virtual void onDisconnect(BLEServer* pServer) override;
                             ^~~~~~~~~
/Users/sairanga/Documents/Arduino/libraries/ESP32-USB-TO-BLE-master/BleDevice.h:31:16: error: 'virtual void BleDevice::onConnect(int*)' marked 'override', but does not override
   virtual void onConnect(BLEServer* pServer) override;
                ^~~~~~~~~
/Users/sairanga/Documents/Arduino/libraries/ESP32-USB-TO-BLE-master/BleDevice.h:32:16: error: 'virtual void BleDevice::onDisconnect(int*)' marked 'override', but does not override
   virtual void onDisconnect(BLEServer* pServer) override;
                ^~~~~~~~~~~~
Multiple libraries were found for "BLEHIDDevice.h"
 Used: /Users/sairanga/Documents/Arduino/libraries/BLE
 Not used: /Users/sairanga/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/BLE
exit status 1
Error compiling for board ESP32 Dev Module.

these are all the errors 
can you help me with these

peleccom, 2024-08-28:
Try to install esp32 board version 2.0.17 instead of 3 in arduino

rakshitbharat, 2024-12-02:
#14 try this PR

rakshitbharat, 2024-12-03:
> Try to install esp32 board version 2.0.17 instead of 3 in arduino

i can confirm this works
[image]

closed 2024-12-16 by rakshitbharat