jmdmahdi/ESP32-USB-TO-BLE #11: Not working with mechanical keyboard.

trungtp147, 2024-10-12:
This code works effectively with regular usb keyboards like HP, Dell, Lenovo, etc., but it does not work with mechanical keyboards (tested with two brands: E-Dra and Dare-U).

rakshitbharat, 2024-12-03:
# ESP32 USB Host to BLE Keyboard - Issue Analysis

## Keyboard Specifications
- Product: SINO WEALTH Gaming Keyboard
- VID: 0x258a
- PID: 0x002a
- Speed: Full-speed (12 Mb/s)
- Current Required: 500mA

## What Went Wrong

### 1. Signal Level Issues
- ESP32 internal pull-up (~45K ohm) too weak for USB signaling
- USB requires specific resistance values:
  * D+ needs 1.5K ohm pull-up for full-speed detection
  * D- needs 15K ohm pull-down for proper signaling
- Current setup has incorrect signal levels preventing device detection

### 2. Power Requirements
- Mechanical keyboard needs 500mA
- ESP32 USB host might not provide enough power
- Unstable power can prevent proper enumeration

### 3. Speed Detection Failure
- Without proper 1.5K pull-up on D+:
  * ESP32 can't detect full-speed device
  * Keyboard speed negotiation fails
  * Device remains undetected

### 4. Signal Integrity
- Long wires or poor connections
- Missing proper pull-up/down resistors
- Incorrect voltage levels on D+/D-

## Why It Didn't Work

1. **Device Detection**
   - ESP32 couldn't properly detect keyboard because:
     * Missing 1.5K ohm pull-up on D+
     * Missing 15K ohm pull-down on D-
     * Internal resistors too weak for USB spec

2. **Power Issues**
   - Mechanical keyboard needs stable 500mA
   - USB power must be properly supplied

3. **Signal Requirements**
   - USB full-speed devices need specific voltage levels
   - Internal pull-up/down can't meet USB specifications
   - Signal timing affected by incorrect resistance values

## Required Fixes

1. **Add External Resistors**
   ```
   D+ (GPIO 16) ----[1.5K ohm]---- 3.3V
   D- (GPIO 17) ----[15K ohm]----- GND
   ```

2. **Power Supply**
   - Ensure stable 5V supply
   - Capable of providing 500mA

3. **Clean Connections**
   - Short, direct wires
   - Proper ground connection
   - Good quality USB socket

## Expected Behavior After Fix
1. ESP32 detects keyboard connection
2. Full-speed (12 Mb/s) negotiation succeeds
3. Keyboard enumeration completes
4. Device appears as BLE keyboard
5. Keystrokes transmitted properly

Save this analysis for future reference when implementing the fixes.

rakshitbharat, 2024-12-03:
# Final Fix for ESP32 USB Host

## Hardware Requirements
1. **Essential Components:**
   - USB Type-A Female Socket
   - Two specific resistors:
     * 1.5K ohm resistor (1,500 ohm)
     * 15K ohm resistor (15,000 ohm)
   - Good quality jumper wires
   - Breadboard or PCB

2. **Connection Layout:**
   ```
   USB Socket     ESP32           Resistors
   Pin 1 (VBus) -- 5V
   Pin 2 (D-)   -- GPIO 17 ------|
   Pin 3 (D+)   -- GPIO 16 ------|
   Pin 4 (GND)  -- GND           |
                                 |
   3.3V ----[1.5K ohm]---- D+ connection
   GND  ----[15K ohm]----- D- connection
   ```

## Code Configuration
Keep the current code but add these key elements:

1. **Pin Setup:**
   ```cpp
   // Configure pins as floating since we have external resistors
   gpio_set_direction((gpio_num_t)16, GPIO_MODE_INPUT_OUTPUT);
   gpio_set_direction((gpio_num_t)17, GPIO_MODE_INPUT_OUTPUT);
   gpio_set_pull_mode((gpio_num_t)16, GPIO_FLOATING);
   gpio_set_pull_mode((gpio_num_t)17, GPIO_FLOATING);
   ```

2. **Power Considerations:**
   - Use powered USB hub if keyboard needs more power
   - Or provide external 5V power supply

## Important Notes
1. No shortcuts on resistor values - they are mandatory
2. Keep connections short and clean
3. Good ground connection is essential
4. Double-check all connections before powering up
5. Use multimeter to verify resistor values

This setup will work for:
- Your SINO WEALTH keyboard
- Other USB keyboards
- Most USB HID devices

rakshitbharat, 2024-12-16:
information about this is how added so closing it

closed 2024-12-16 by rakshitbharat