jmdmahdi/ESP32-USB-TO-BLE #17: ESP32-S3-WROOM-1-N16R8 Stuck in boot loop

Moose-Gaming, 2024-12-19:
Hi,

I tried to set up everything according to the instructions. I couldn't figure out how to install ESP32 board support package - that is the only thing in my set up that is different from the instructions. I have the latest Arduino IDE, libraries, etc. The `platformio.ini` looks like this: 
```
; Flash: 16MB QD, PSRAM: 8MB OT
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200

build_unflags = 
    -std=gnu++11

build_flags = 
    -std=gnu++17
    -DCORE_DEBUG_LEVEL=5
    -I include
    -I src
    -I .pio/libdeps/esp32dev/NimBLE-Arduino/src
    -I ${platformio.packages_dir}/framework-arduinoespressif32/tools/sdk/esp32/include

lib_deps =
    https://github.com/T-vK/ESP32-BLE-Keyboard
    https://github.com/T-vK/ESP32-BLE-Mouse
    tobozo/ESP32-USB-Soft-Host
    h2zero/NimBLE-Arduino @ ^1.4.1

lib_ldf_mode = deep
build_src_filter = +<*> -<.git/> -<.svn/>

monitor_filters = esp32_exception_decoder

board_build.arduino.memory_type = qio_opi
board_build.flash_mode = qio
board_build.psram_type = opi
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.extra_flags = 
  -DBOARD_HAS_PSRAM
```

The extra set up is based on this: https://github.com/sivar2311/ESP32-S3-PlatformIO-Flash-and-PSRAM-configurations?tab=readme-ov-file#esp32-s3-wroom-11u-n16r8

After flashing the ESP is in a constant boot loop with this as the serial output
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420534ba
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[   149][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
USB#0 (pins 16 17) is OK!
READ_BOTH_PINS = 0000
READ_BOTH_PINS = 0000
READ_BOTH_PINS = 0000
READ_BOTH_PINS = 0000
cpu freq = 240 MHz
TIME_MULT = 51 
D=   0 4800 bits in 1003.000000 uSec 4.785643 MHz  6 ticks in 1.253750 uS
D= 127 4800 bits in 4458.000000 uSec 1.076716 MHz  6 ticks in 5.572500 uS
D=  63 4800 bits in 2581.000000 uSec 1.859744 MHz  6 ticks in 3.226250 uS
D=  95 4800 bits in 3517.000000 uSec 1.364799 MHz  6 ticks in 4.396250 uS
D=  79 4800 bits in 3048.000000 uSec 1.574803 MHz  6 ticks in 3.810000 uS
D=  87 4800 bits in 3216.000000 uSec 1.492537 MHz  6 ticks in 4.020000 uS
D=  83 4800 bits in 3128.000000 uSec 1.534527 MHz  6 ticks in 3.910000 uS
D=  85 4800 bits in 3177.000000 uSec 1.510859 MHz  6 ticks in 3.971250 uS
D=  86 4800 bits in 3189.000000 uSec 1.505174 MHz  6 ticks in 3.986250 uS
D=  86 4800 bits in 3196.000000 uSec 1.501877 MHz  6 ticks in 3.995000 uS
TRANSMIT_TIME_DELAY = 88 time = 3.995000 error = -0.125003% 
USB#1 (pins 22 23) is OK!
E (87) gpio: gpio_set_direction(275): GPIO number error
E (92) gpio: gpio_set_level(227): GPIO output gpio_num error
E (97) gpio: gpio_set_direction(275): GPIO number error
E (102) gpio: gpio_pulldown_en(110): GPIO number error
E (107) gpio: gpio_set_direction(275): GPIO number error
E (112) gpio: gpio_set_level(227): GPIO output gpio_num error
E (118) gpio: gpio_set_direction(275): GPIO number error
E (123) gpio: gpio_pulldown_en(110): GPIO number error
READ_BOTH_PINS = 0000
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42004245  PS      : 0x00060c30  A0      : 0x82002931  A1      : 0x3fcebb40  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3fca5f24  A5      : 0x3fca4cf8  
A6      : 0x3fca4d10  A7      : 0x00000001  A8      : 0x3c067b70  A9      : 0x60004000  
A10     : 0x00000000  A11     : 0x00400000  A12     : 0x3fca5d0c  A13     : 0x3fca4f18  
A14     : 0xfffffdff  A15     : 0xbfc00018  SAR     : 0x00000009  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x42004242:0x3fcebb40 0x4200292e:0x3fcebbb0 0x42002a00:0x3fcebc00 0x42002720:0x3fcebc30 0x420178ca:0x3fcebc50
````

As I'm a beginner ESP32 user (only experience is ESPHome) I have no idea what this means. 

Thanks in advance!

rakshitbharat, 2024-12-30:
Thanks for opening issues, I will look into it soon

rakshitbharat, 2025-01-02:
> Hi,
>
> I tried to set up everything according to the instructions. I couldn't figure out how to install ESP32 board support package - that is the only thing in my set up that is different from the [...]

Can you please test #19 
Let me know if this new changes works for you

glasscake, 2025-01-05:
Im also working with the USB C clone of the S3-Devkit see: https://github.com/rtek1000/YD-ESP32-23/tree/main

Moose-Gaming, 2025-01-07:
Hi,
sorry about this, but I'm pretty new to GitHub and don't really know how to test [#19](https://github.com/jmdmahdi/ESP32-USB-TO-BLE/pull/19). I don't know how to download this :(.

rakshitbharat, 2025-01-09:
let me know when you're able to do so and test it on your hardware 
thanks

we will wait for your reply

rakshitbharat, 2025-01-12:
closing because of no reply

closed 2025-01-12 by rakshitbharat