jmdmahdi/ESP32-USB-TO-BLE #18: Does This code work with mechanical keyboard?

jahidulislamjahid, 2024-12-22:
Hi there. I flashed an esp wroom32 (esp32 DevKit v1) about 3 weeks ago. code flashed successfully and I can connect it to my computer as an Esp32 keyboard but the issue is my keyboard. I have a mechanical keyboard. and I saw about a week or two ago a solution for the mechanical keyboard issue which referred to adding pull-up and pull-down resistors to the board. but lost the instruction and now I cannot find it. also, I see the project has been updated a few times. so I want to know is the updated code works with the mechanical keyboard as it is? or do I need the registers? If resistors are needed than can you kindly give me the instructions on how to do that?
and Thanks in advance.

rakshitbharat, 2024-12-30:
Check read me properly and all closed issues you will find your answer,

thanks

closed 2024-12-30 by rakshitbharat

jahidulislamjahid, 2024-12-30:
Hi, I've got the instructions in a closed issue. but I'm not sure exactly where and in which file should I put those gpio codes. can you just tell me that. I have the required resistors. Just tell me where and in which file should I put these code. 
```// Configure pins as floating since we have external resistors
gpio_set_direction((gpio_num_t)16, GPIO_MODE_INPUT_OUTPUT);
gpio_set_direction((gpio_num_t)17, GPIO_MODE_INPUT_OUTPUT);
gpio_set_pull_mode((gpio_num_t)16, GPIO_FLOATING);
gpio_set_pull_mode((gpio_num_t)17, GPIO_FLOATING);```
And I am using this board.
![20241230_002451_lmc_8 4](https://github.com/user-attachments/assets/c0a7829e-16b4-436f-bd5a-14bc87681d03)
![20241230_002549_lmc_8 4](https://github.com/user-attachments/assets/ebef665f-8abf-4922-a5e6-2189480821c6)